stuartpittaway/diyBMSv4Code #114: ESP8266 "Press SPACE BAR to enter terminal based configuration" not working

dakoal, 2021-06-22:
On ESP32 the "Press SPACE BAR to enter terminal based configuration...." is working fine.
But with latest release downloaded from github for ESP8266 it's not working.

It's always jumping to "skipped" after timeout.

Te code portion I think is:

//Allow user to press SPACE BAR key on serial terminal
  //to enter text based WIFI setup
  SERIAL_DATA.print(F("\r\n\r\n\r\nPress SPACE BAR to enter terminal based configuration...."));
  for (size_t i = 0; i < (3000 / 250); i++)
  {
    SERIAL_DATA.print('.');
    while (SERIAL_DATA.available())
    {
      int x = SERIAL_DATA.read();
      //SPACE BAR
      if (x == 32)
      {
        TerminalBasedWifiSetup(SERIAL_DATA);
      }
    }
    delay(250);
  }

stuartpittaway, 2021-06-22:
Have you used this on the esp8266 previously?

The code was merged from 32 to 8266 to try and maintain some resemblance of similarity, but I didn't publish that it was a working feature.

dakoal, 2021-06-22:
Ok, I didn‘t know that.

Only saw the line coming on booting.
Sorry for disturbing

stuartpittaway, 2021-06-22:
I'm not saying it shouldn't work, but its probably not been tested/used!