stuartpittaway/diyBMSv4Code #115: ATTINYCellModule common wire problem cause measure error

almol, 2021-06-28:
As described in [https://community.openenergymonitor.org/t/diybms-v4/11292/3849](url) error on voltage measurements will be produced when contigous module is balancing because a common wire cabling between modules and cell batteries.
This causes the module to also begin to balance. Because it measures a higher voltage than the real one, and a chain effect occurs.

To fix it, it would be necessary that no voltage be measured if the adjoining module is balancing. This could be solved in several ways, for example taking advantage of the moment of data communication to stop balancing and perform the measurement.

We cannot assume that common wiring is not present in all battery configurations.

stuartpittaway, 2021-06-28:
That's for submitting this.

Diybms works very similar to other BMS systems with a balance function, such as Batrium.

If the modules were limited to only balance one at a time, or not when a neighbor was balancing then it would take a very long time to balance a pack.

Additionally, a cell could drift over voltage and no way to prevent this occuring.

almol, 2021-06-30:
Hi.
The idea is to stop balancing very few milliseconds.
If no neighbor is detected, no comm or other issue, normal balancing will be performed.