stuartpittaway/diyBMSv4Code #117: MQTT compatibility with HomeAssistant

vajpeters, 2021-07-10:
Since HomeAssistant is getting a defacto standard, it would be nice to create a MQTT compatibility mode with HomeAssistant.

stuartpittaway, 2021-07-10:
What's needed to make it work?

vajpeters, 2021-07-10:
It's pretty well documented here:
https://www.home-assistant.io/docs/mqtt/discovery/
Mainly it's about naming and structure conventions. Would this be possible?

shampeon, 2021-09-11:
Just another vote for adding the necessary auto-discovery topics. This would make it incredibly easy to incorporate a diyBMS battery into Home Assistant's new Energy platform.

stuartpittaway, 2021-09-11:
A battery or BMS doesn't appear to be in the list of supported discoverable devices. Is the document out of date?