stuartpittaway/diyBMSv4Code #118: Random reboots through out the day

magpern, 2021-09-16:
I have random reboots of the ESP32, several times a day.
New to the setup is the shunt, but also influx and mqtt.
My guess is it is related to influx

Attached is a stacktrace
Line 36568

[diyBMSController.txt](https://github.com/stuartpittaway/diyBMSv4Code/files/7180711/diyBMSController.txt)

atanisoft, 2021-09-16:
```
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x4011bc57 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40090d8c:0x3ffe0ee0 0x40091009:0x3ffe0f00 0x4011bc57:0x3ffe0f20 0x401495be:0x3ffe0f50 0x40149645:0x3ffe0f70 0x400e552a:0x3ffe0f90 0x401487a0:0x3ffe0fb0 0x40092cf6:0x3ffe0fe0

Rebooting...
ets Jun  8 2016 00:22:57
```

The assert is coming from [this](https://github.com/espressif/esp-lwip/blob/2.0.3-esp/src/core/tcp.c#L779) line, it looks like a data overrun (too much data being RX or TX?).

@magpern Can you decode the rest of the backtrace to determine which part of diyBMS (if any) was involved?

magpern, 2021-09-16:
> @magpern Can you decode the rest of the backtrace to determine which part of diyBMS (if any) was involved?

I probably can, but how?

Right now I am just dumping what ever I get från the ESP32 through a python script running on a raspberry pi

atanisoft, 2021-09-16:
You should be able to use decoder.py from https://github.com/me21/EspArduinoExceptionDecoder as:

`decoder.py -p ESP32 -t ~/.platformio/packages/toolchain-xtensa32 -e {path to the diyBMS elf file} {file containing the backtrace}`

Also note that this issue should really be over on https://github.com/stuartpittaway/diyBMSv4ESP32 (which @stuartpittaway likely can move it there)