stuartpittaway/diyBMSv4Code #124: Feature discussion: save/restore all settings via browser

datenheim, 2022-06-01:
While playing around with the system (Release-2022-05-24-08-14) and starting to modify some code lines (related to MQTT) I noted that all settings (module settings, rules, integrations etc.) are gone if a new firmware is uploaded to the controller (D1 mini in my case).

While this might be unavoidable due to ESP8266 and likely ESP32 it came to my mind that it would be great to just "download" the settings to some kind of file (maybe JSON).
Than after flashing the new binary one could restore all settings by "uploading" the formerly mentioned file.

Is there another way or workaround to achieve this?
How could that be implemented?

atanisoft, 2022-06-01:
The esp32 controller has support for storage of configuration data on an SD card which is checked upon startup to load configuration data from.

sglux, 2022-06-02:
Thanks for that hint. I'll take a look, maybe it could be implemented here too.
Couldn't SPIFFS (or whatever it's called today) used for that?

atanisoft, 2022-06-02:
SPIFFS is stored in flash, when flashing via the single binary method it will wipe flash prior to writing the new image so the SPIFFS data would be lost.

stuartpittaway, 2022-06-08:
You could simply download the file via a web browser and then upload after the upgrade.

datenheim, 2022-06-09:
> You could simply download the file via a web browser and then upload after the upgrade. No SD card required then.
Hi Stuart,
I did not find any option in the GUI of the ESP8266 controller (Release-2022-05-24-08-14).
Do I miss something?

stuartpittaway, 2022-06-09:
No you didn't miss anything, I was just mentioning that an SD card wasn't mandatory, other methods could be written.