stuartpittaway/diyBMSv4Code #127: SOC with PZEM-017

Michi202020, 2022-10-09:
Hello again,

it is possible that we can the soc calculation direct on the ESP32 controller for the PZEM-017. He can not make it self. What he can is Voltage, amps and Watts and kwh sent to the controller.

I hope you can give me a answer.

Best Regards
Michael

stuartpittaway, 2022-10-09:
Not with the pzem device, you need to sample the voltage and current hundreds of times per second to get an accurate kWh count.

Michi202020, 2022-10-09:
Hello Stuart,
thats right but the PZEM-017 have a own kwh counter. Is it possible that you ask all 5 sec. the kwh and yous it for soc?
For example after your soc reset (max voltage and current go down) can you scale a battery of 10 kwh and scale it to 0-100%.
You dont need to sample the voltage and current hundreds of times per second.
For the Victron is this a Better calculation for the Work.

Best Regards 
Michael

stuartpittaway, 2022-10-09:
The pzem can't measure direction of current, so you have no idea if it's charging or discharge, so calculation of soc isn't going to work

Michi202020, 2022-10-11:
Hello Stuart,
That's right and a problem. I have found a option to the Pzem-017. The Pzem 025 is Bidirectional and withe display. Is this a option for the soc? 

https://www.kaufland.de/product/442973006/?kwd&source=pla&sid=37703122&gclid=CjwKCAjwqJSaBhBUEiwAg5W9p9MSlaOnIf7LmXLy1hWOr0Kdg4i9cPPCMRCbK0hQmiY2gyDte2o09xoCXaYQAvD_BwE

Michi202020, 2022-10-12:
Hello again, thit was wrong what I sent. He don't have a RS485 but what do you think about This? There you can use this for all Shunt withe 75mV.
The Price is good withe round about 20$

https://www.ivy-emeter.com/html/acenergymeter-show-678.html

Best Regards
Michael

Michi202020, 2022-10-12:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.aliexpress.com/item/1005004659359322.html&ved=2ahUKEwjnvIS0_dn6AhWmW_EDHa7CCs8QFnoECCEQAQ&usg=AOvVaw3vzdVZ5WfsjUMQCg2stEPV

stuartpittaway, 2022-10-12:
Those look good, but I can't see that either do soc measurements.

Michi202020, 2022-10-12:
That's right but is this not easyer that you ask from the meter the kWh and scale this in soc.
For example the battery what do you install is 20 kwh and this is 100 % and the meter is Bidirectional that the kwh go up and down also you can reset this with a command to 0 of the meter. That's almost easyer as you calculate withe amps.