stuartpittaway/diyBMSv4Code #132: Setting 9k6 or 5k baudrate in platformio.ini

chron0, 2022-12-26:
Ahoy, i finally got all the attinys soldered and was trying to compile and flash cellmodule firmware into mine.

Note: There was no info about using an AVRisp2, a search turned up a lot of misleading information out there (platformio forum mostly) while a simple upload_protocol change seems to work without a problem, so if anyone else want to flash the modules with an AVRISPMK2, you can give this a try:

```
--- a/ATTINYCellModule/platformio.ini
+++ b/ATTINYCellModule/platformio.ini
@@ -48,7 +48,9 @@ extra_scripts =
 lib_deps =
       https://github.com/mike-matera/FastPID.git
       https://github.com/stuartpittaway/SerialEncoder
-upload_protocol = usbasp
+
+;upload_protocol = usbasp
+upload_protocol = stk500v2
 ; Fuses E:F4, H:D6, L:62
 ; lfuse = 0110 0010 = CKDIV8 (enabled) & Calibrated Internal 8MHz Oscillator
 ; hfuse = 1101 0110 = EESAVE & 1.8V BOD detection level
```

Might be worth to be added to readme.

Question: I did not find any hint about compiling a V440 firmware with different baudrate in the docs. 
I assume it is probably as simple as adding another env to platformio.ini like this?

```
[env:V440_9K6]
build_flags=-DDIYBMSMODULEVERSION=440 -DMV_PER_ADC=2.00 -DINT_BCOEFFICIENT=3950 -DEXT_BCOEFFICIENT=3950 -DLOAD_RESISTANCE=3.30 -DBAUD=9600
board_fuses.lfuse = 0b01101100
board_fuses.hfuse = 0b11010110
board_fuses.efuse = 0b11110100
upload_flags =
  -Pusb
  -Ulfuse:w:0b01101100:m
  -Uhfuse:w:0b11010110:m
  -Uefuse:w:0b11110100:m
```

stuartpittaway, 2022-12-26:
Hello, absolutely no need to use platform IO unless you plan to make code changes. 

Just used the precompiled files from GitHub releases, and follow the instructions!

chron0, 2022-12-26:
Whats the point of open-source then, if I would only download binary files someone else has to supply :) And yes, i might play around with the code a bit in the future as well - the main reason I picked diyBMS IS that it is open-source and gives us that option. And that means learning and understanding more about it and adding a little to the documentation for others to be able to do it as well :)

So, again, -DBAUD=9600 enough for different baudrates or are other changes needed?

stuartpittaway, 2022-12-26:
Well if you are using platform.io, use the default 'compile all' option and it will spit out all the binary files for all modules at all supported baud rates.

chron0, 2022-12-30:
ah, thx, I'll try it this way then.

closed 2023-01-01 by chron0