stuartpittaway/diyBMSv4Code #133: Programming Module V4.5

Solarbot, 2023-03-23:
Would it be possible to include information on programming the latest V4.5 module which doesn't have an ISP header?  Many thanks.

atanisoft, 2023-03-23:
I believe it is the same process as the external current shunt with UPDI.

Solarbot, 2023-03-24:
Hi both, thanks for responding.

I've looked up UPDI programming which seems pretty neat - am I correct in thinking that the V4.5 module cannot therefore be programmed by the DIYBMS controller?  If so, may I ask if I need to compile and upload or is there a precompiled version availble?  Also, happy to say the ATTINY seems to be avaible once more :-)

stuartpittaway, 2023-03-24:
No guide on this one I'm afraid - use the 5K if you are starting out - it refers to the communication speed between the modules and controller.

Do some google searching on UPDI programming - its quite simple, but a lot do struggle with it.

Solarbot, 2023-03-24:
The advantage of working it out for myself is that new stuff gets learnt :-)

Thanks for the pointers 👍 

Cheers

atanisoft, 2023-03-24:
@stuartpittaway https://github.com/adafruit/Adafruit_AVRProg or https://gitlab.com/bradanlane/portaprog/-/tree/master/ might provide the functionality. I didn't have much luck with Adafruit_AVRProg on a standalone ESP32 setup, but portaprog might be usable. The key will be disabling all other functionality of the BMS during flashing to maximize the time allotted to UART interaction.

Solarbot, 2023-10-16:
Hi Stuart,

 

I’m just looking at this again and found BitLuni’s guide to using an Arduino to upload using UPDI along with ATtinyCore by SpenceKonde which looks like a nice option, links below.

 

I haven’t had any luck as yet using your files on github all saved in one folder with the file main.cpp saved as an ‘ino’ file.  I don’t have an option to select 2MHZ as the clock speed for the MCU selected as ATTINY841 (no bootloader) and get the compile error below.

 

Am I barking up the wrong tree?  I’ve also looked at using an FTDI serial board and XLOADER but there is no option to select ATTINY841 as the MCU.

 

A bit stuck… any pointers much appreciated. 

 

Cheers,

 

Cam

 

 

https://www.youtube.com/watch?v=AL9vK_xMt4E <https://www.youtube.com/watch?v=AL9vK_xMt4E&t=463s> &t=463s 

 

https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/Installation.md

 

Compile error:

 

In file included from C:\Users\Me\Documents\Arduino\DIYBMS45\main.cpp:46:0:

diybms_attiny841.h:18:2: error: #error Processor speed should be 2Mhz

#error Processor speed should be 2Mhz

  ^~~~~

main.cpp:49:2: error: #error Expected DIYBMSBAUD define

#error Expected DIYBMSBAUD define

  ^~~~~

main.cpp:53:2: error: #error Expected SAMPLEAVERAGING define

#error Expected SAMPLEAVERAGING define

  ^~~~~

In file included from C:\Users\Me\Documents\Arduino\DIYBMS45\main.cpp:57:0:

defines.h:15:10: fatal error: EmbeddedFiles_Defines.h: No such file or directory

#include "EmbeddedFiles_Defines.h"

          ^~~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

#error Processor speed should be 2Mhz

stuartpittaway, 2023-10-17:
Hi @Solarbot 

Use the pre-compiled firmware/code directly from GITHUB Releases.

Additionally, use a simple USB Serial adapter as an UPDI programmer, you don't need anything fancy, diode and couple of resistors.  https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md#serial-adapter-requirements

After this, AVRDUDE command line tool will program the chips

Alternatively, you can use a spare Arduino Uno like in this guide - https://daumemo.com/diy-updi-usb-programmer-which-can-be-made-with-cheap-hardware/

And finally, another option is to use the Microchip python code...
https://github.com/microchip-pic-avr-tools/pymcuprog

Solarbot, 2023-10-17:
Hi Stuart,

 

Perfect, thank you – AVRDUDESS proved to be the key for me in the end.

 

Much appreciated as is all your work – I’m slowly working towards building a battery but at the rate I’m going you will have a whole new ecosystem up and running by then :-)

 

Cheers.

 

https://github.com/microchip-pic-avr-tools/pymcuprog

closed 2023-10-17 by stuartpittaway